re3Dprinting/touchscreen #36: Manual extrusion not working

jim-thompson, 2020-01-30:
Heat up extruders then go to Control screen. Touching +E should extrude 10 mm of filament, but does nothing.

jim-thompson, 2020-02-22:
Found and fixed code selecting the wrong extruders. Have not tested to see if it's fixed, but moving to lower priority.

plloppii, 2020-08-08:
Witness when I select manual extrusion to 10mm, I see about 500mm- 1000mm extrusion, and needed to reboot to stop extrusion.

re3Dprinting, 2020-08-27:
version b 1.0.0
manual extrusion is working on both tool heads. I think there is an issue with incremental vs absolute position. I told the right extruder to move 10mm of filament and it moved 10x too far. 

Also the extrusion rate is really slow - we may want a way to adjust the extrusion rate