re3Dprinting/touchscreen #49: Visual: fix duplicate tags on menu screen

Alemontfer1992, 2020-02-17:
We should delete the "re:3Display" on the upper left corner of the menu screen because we already have it on the opposite corner. 

[image]

jim-thompson, 2020-02-17:
The one in the lower corner has been replaced by an information string listing things like the touchscreen's IP address and the Git commit it was built from.

Alemontfer1992, 2020-02-17:
I think we should just remove it. It's redundant: we know it is a display and the re:3D logo is there, branding this display as the re:3D display

plloppii, 2020-08-04:
fixed in latest devs, devel/1.0.28

closed 2020-08-04 by plloppii