re3Dprinting/touchscreen #51: Visual: Icons are a bit confusing

Alemontfer1992, 2020-02-17:
There are a couple of icons that are just really confusing. The first time I looked at these screens, this is what I instantly translated each icon into:

[image]

[image]

In that same order, this is what I would expect from those categories:
- Print file: print my file
- Movement: move the bed and print head
- Temperature: preheat extruder/bed
- Settings: all options for printer tuning/calibration 

- Connectivity: connect to wifi, printer, etc.
- Murder: explode printer

jim-thompson, 2020-02-17:
LOL! Not murder. That's the icon for the serial port setup screen. Serial port / serial killer. So just a little murdery.

jim-thompson, 2020-02-17:
But I totally agree that the icons aren't very good. Matthew and I have talked about doing a complete visual redesign, and that would rely heavily on you!

Alemontfer1992, 2020-02-17:
I would be more than happy to help you with the icons and visual aspect of it.

plloppii, 2020-08-04:
Patched during the latest devs, updating the icons.

closed 2020-08-04 by plloppii