re3Dprinting/touchscreen #52: Interface could be smoother

Alemontfer1992, 2020-02-17:


chief-hacker, 2020-02-23:
> Suggestions:
> 
> I don't know if any of these are possible but I'll just go ahead and share all my suggestions:
> 
> **Print file screen**
> 
> * Change SD, USB, Local to icons
> * Display files as icons with file name instead of a list.

Why Icons? I prefer reading all my folder contents as lists unless they are photos

> * When you click file icon, it would take you to a separate screen with the name of the print and 2 options: Print | Cancel
>   - If you click "Print", it takes you immediately to the "print status" screen
>   - If you click "Cancel", it takes you back to the file options
> * Once there's a file printing, can we hide file options? I'm already printing one, so I don't need to see that.
> * If you click the print file icon while already printing something, we could display this message "You are currently printing 'file name', do you wish to cancel?" followed by 2 options: Yes | No
> 
> **Print file status screen**
> 
> * Display only the immediate information we need to know
>   -  File name at the top

Could the filename be listed in the status bar at the bottom of the screen?

>   -  Extruder temperature
>   -  Bed temperature
>   -  Current xyz position
>   -  Print progress bar
>   -  Print time
>   -  Stop print icon. If you click, it gives you a pop up warning: Are you sure you want to cancel? Yes | No. Clicking "Yes" takes you to print file options screen again
>   - Pause print icon switch that turns into resume print icon
>   - Tune print icon. On click, takes you to babystepping, temperature, flowrate...
>   - Back & home buttons
>   - Error warning icon with code. On click, takes you to the error(s) with a standard description of what it means and even maybe a link to any resources we have that could help you fix it.

Alemontfer1992, 2020-03-03:
The icon thing is just a preference thing. I find it easier for browsing files but it doesn't really need to change. 

The file name can be listed at the bottom as long as it is easy to locate and read.