re3Dprinting/touchscreen #53: Issue with homing xy

Alemontfer1992, 2020-02-17:


jim-thompson, 2020-02-17:
I believe if you follow home XY with a home Z, it should take care of that.

jim-thompson, 2020-02-17:
Is this different behavior than what you get if you home XY using Viki?

Alemontfer1992, 2020-02-17:
Yup. When you home XY using Viki, I believe only XY move and the Z doesn't lower at all. I just found it weird because I home Z first, then XY and then I had to home Z again. This way, it's sort of forcing you to home all but in a 2 step process.

jim-thompson, 2020-02-19:
OK, I'll check it out. Thanks for the feedback.

chief-hacker, 2020-02-23:
> Yup. When you home XY using Viki, I believe only XY move and the Z doesn't lower at all. I just found it weird because I home Z first, then XY and then I had to home Z again. This way, it's sort of forcing you to home all but in a 2 step process.

Possible solution - Have a "Home All" button that will allow you to home everything with one button push.

jim-thompson, 2020-02-23:
We have a Home All button, next to the Motor Off button. It's not labeled.