re3Dprinting/touchscreen #78: Touchscreen crash on Babystepping

re3dCharlotte, 2020-04-13:
On Gb3-525, I was approximately 2 mins into a print. (April 13, 5:48pm) Touched the baby stepping up arrow and the touch screen went black then restarted (showed logo screen) . The print stopped.
Reconnected, restarted the print, touched babystepping up arrow, and the same issue happened again.
I added a log marker both times.

jim-thompson, 2020-04-15:
Found the problem. I'm going to install a fix Thursday morning (4/16), so if you could please try babystepping again sometime after that, and please let me know how it goes?

re3dCharlotte, 2020-04-16:
Thanks Jim. I won't be there until this evening, so I've included Aubrey
and Matthew in case they want to test out babystepping on the touchscreen
before that.

[image:
https://re3d.org/covid-19-update-operations-serving-educators-joining-the-fight/]
<https://re3d.org/covid-19-update-operations-serving-educators-joining-the-fight/>
<https://re3d.org/careers/>

jim-thompson, 2020-04-16:
Fix is not active an all the printers. I'll be in early Friday to update the inactive printers.

closed 2020-04-17 by jim-thompson