re3Dprinting/touchscreen #90: Print Not Stopping

SamReeve, 2020-07-24:
I used the touchscreen to start a print off the SD card. I meant to start it off the USB, but the SD window appeared first and I selected the wrong print. When I realized I started the wrong print, I used the touchscreen to stop it. The status changed to canceling and the touchscreen acted as if the print was stopped, but the printer kept printing for at least 5 min. The printer was printing a series of small triangles, so the buffer should have taken less than a minute to clear. When I used the viki to stop the print (after telling it to stop on the touchscreen), it stopped immediately. The touchscreen was stuck trying to cancel the print and wouldn't let me interface with the printer. Reconnecting to the printer reset the status and let me get the printer working again. If you want to look at the logs it should be just after 7am 7/24/2020 on GB3+525.

re3Dprinting, 2020-07-27:
Samantha,

Can you replicate this with the latest version 1.0.26 and let me know the results.

plloppii, 2020-07-27:
I am currently looking into this, I have reproduced this on my end as well.

re3Dprinting, 2020-08-28:
Version b 1.0.0
verified print not stopping when printing from SD card