re3Dprinting/touchscreen #108: Zits and Blobs Through Printing

plloppii, 2020-08-08:
Big Blocker that affects both user experience and functionality of the touchscreen. 

"During the print the machine exhibited the frequent micro-pauses we have come to find when using the touchscreen. I unplugged the USB cable from the touchscreen and all of a sudden 
the printing smoothed out and the machine movements were smooth."

Theories below:
https://community.octoprint.org/t/blobs-on-the-surface-of-prints/17631/5
https://community.octoprint.org/t/the-ongoing-usb-conspiracy-theory/6341

Things to test:
Decreasing print speed
Decreasing jerk and acceleration
Testing touchscreen on 32bit board
Different USB cable

Interesting plugin that compresses gcode files:
https://community.octoprint.org/t/new-plugin-anti-stutter-need-testers/18077

plloppii, 2020-08-08:
The logging system that was implemented is still dumping print logs into a file system. As documented in Octoprint's code, this can significantly reduce performance. This needs to be turned off to allow the software to function optimally.

plloppii, 2020-08-08:
Updated list of things to test:
Varying print speed
Varying jerk and acceleration 
Isolating Touchscreen prints from Stock Octoprint instance print
Octoprint instance with anti-stutter plugin

jim-thompson, 2020-08-08:
I tried: (a) turning off logging entirely, and (b) turning off just the serial logging. Neither had an appreciable effect on the zits problem. Turning off logging entirely would leave you with almost no visibility into problems that happen in the field.

Consider an alternative: move to logs onto a ramfs filesystem. The Pi 4 4GB has more than enough RAM to support a ramdisk for logging, which would leave you with logging but onto a very fast filesystem. NB: use ramfs as the backing device type not tmpfs.

re3Dprinting, 2020-08-28:
Update - The micro pause issue is happening even if I am not printing from the touchscreen. The micro pause happens when I am printing from SD and the Viki - with the touchscreen connected via USB. If I disconnect the USB from the touchscreen the printing smooths out.

How does just having the touchscreen connected during a print cause Marlin to exhibit micro pausing?